i-am-shodan/USBArmyKnife #186: Feature Requests: Bluetooth USB Keyboard dongle with keycap record

Onlineraider, 2025-07-26:
## Your Feature:
 I Had the Idea to Use the Bluetooth Module of the Army knife to connect a BT Keyboard to the usbArmy knife which is connected to the PC via USB (or also via Bluetooth If possible) and acts AS a USB Keyboard while recording the pressed Keys on the Keyboard (with delay time etc)

## Your device

```txt
LilyGo T-Dongle S3
```

- I confirm I have read the FAQ: ✔️
- I confirm I am using the latest release: ✔️

- My version number is: 2.1.1

## Additional information

instatablet30-collab, 2025-09-28:
Yes I think he meant that

i-am-shodan, 2026-03-11:
I've looked into it as it is a good idea. The problem is that the Bluetooth stack UAK uses is called NimBLE. This doesn't yet have support for BT HID host.

Writing a reliable implementation is an entire project. One that I'm not keen on taking on.